pekkaroi/bldc-drive #2: Error while compiling with IAR ewarm utils.c

nionut, 2016-08-01:
I get this error at line 11 Error[Og006]: Error in inline assembly: "Error[41]: Bad label" 
The error is in file utils.c  Any clue for debuging?

1   #include "utils.h"
2   #include <stddef.h>
3   #include <stdint.h>
4 
5     void delay_ms(const uint32_t ms)
6    {
7
8        uint32_t ms2 = ms*STM32_CLOCK_HZ / 1000 / STM32_CYCLES_PER_LOOP;
9
10      asm volatile(" mov r0, %[ms2] \n\t"
11               "1: subs r0, #1 \n\t"
12               " bhi 1b \n\t"
13               :
14               : [ms2] "r" (ms2)
15               : "r0");
16    }

pekkaroi, 2016-08-01:
Hi. I have absolutely no experience with IAR, but according to short googling, it might be that IAR does not like the label "1" which is used in the assembly. 

if you try to replace the lines
             "1: subs r0, #1 \n\t"
             " bhi 1b \n\t"

with
             "loop: subs r0, #1 \n\t"
             " bhi loop \n\t"

it might work. At least it compiles with GCC. Unfortunately I have no hardware in my hands right now to test the change. But could you check if this helps?

nionut, 2016-08-01:
I think that I need to instal Eclipse :-(
Error[Og006]: Error in inline assembly: "Error[54]: Expression can not be forward" 
Error[Og006]: Error in inline assembly: "Error[54]: Expression can not be forward"

pekkaroi, 2016-08-01:
Ok, it really seems that IAR handles the asm() keyword differently to GCC. :( I'll try to find a way to implement the delay_ms in a better way without using assembly. I think the SysTick is the right way to go, I just need to learn how to use it.

pekkaroi, 2016-08-01:
I pushed a new version of utils.c. It is now implemented with systick and no assembly. You can try that. I'm afraid this is not the only problem with IAR though, but let's see?

nionut, 2016-08-02:
inline uint32_t millis (void)
{
  return ticks;        //ERROR
}

Error[Pe1031]: an entity with internal linkage cannot be referenced within an inline function with external linkage

pekkaroi, 2016-08-02:
Crap.. This is quite difficult as I cannot test it with IAR. :) It seems to do something with inline support on IAR. I have no clue if the inline declaration is even useful there, I followed an example on web on implementing the Systick delay quickly. You can try it out by simply removing the inline keyword..

nionut, 2016-08-02:
Nop a tried that and i get 101 Errors. IAR gives me headache.

pekkaroi, 2016-08-02:
Right.. I doubt it is the IAR that is causing the problem, it's just my lacking skill in software, the code is something that barely compiles in GCC with luck. :) Sadly I don't have a Windows computer to try with, it seems that there would be a trial version of IAR EWARM available..

nionut, 2016-08-02:
If You whant to install the trial version I can provide the activator serial for full

pekkaroi, 2016-08-03:
I installed the IAR on my girlfriends laptop and I think I made this compile now! (5 warnings are left) I had some serious stupidness on the header files, I'm surprised that GCC is so forgiving that it compiled this. I had some trouble with ST-Link on Windows, so I haven't tried the IAR image on hardware, but at least it compiles.

pekkaroi, 2016-08-03:
Hmm.. After getting the ST-Link to work, I have to say that this is more complicated than I thought. Now I am able to somehow start to run and debug the program, but as soon as first interrupt fires, I get a HardFault. And as I understand it, there is now something mixed with the startup code provided by IAR and the startup code included in my project. I am sorry to say, but I don't think I have enough interest to learn enough about IAR and it's proprietary startup code to really make this work.. :(

nionut, 2016-08-03:
Watch out for the optimizations level under Project-> Options->C/C++ Compiler->Optimizations  it has to be set to None
Then you have to write the path to the included files of your files not IAR in Project-> Options->C/C++ Compiler->Preprocessor->Additional include directories(one per line)
\src
\include
\system\include\cmsis
\system\include\stm32f1-stdperiph

And try to write USE_STDPERIPH_DRIVER
Project-> Options->C/C++ Compiler->Preprocessor->Defined symbols(one per line)

Make sure you select ST STM32F103C8 under Project-> Options->General Options->Target->Device