pekkaroi/bldc-drive #6: schematic&layout

Virtex7, 2017-02-20:
Hi there!

I have taken a look at your schematic and layout, and have seen that you have created it in a kicad version older than 4.0. so i have forked the design and am actually migrating the project to kicad 4.0 and the new symbols. 
if you like to merge it, i will create a pull request once i'm ready with the migrating process.
additionally i have deleted the older versions of schematic and layout, because there is git history and if you like you can pull older versions from there.

please do not migrate at the moment, because there is still work to do and migrating now will lead to more work for you :D

a list of parts and values that you have used in your prototypes would be neat, because they are not mentioned in the schematic...

sry. for something like fucked up grammar or missuse of words, I am from germany and so german is my first and main language...

pekkaroi, 2017-02-20:
Hi, 

I think I have used quite recent git version of kicad, but I am not sure. I'll gladly accept a pull request if you get migration done. I have been working with revision 3 of the controller part. I'll publish that and also try to add component values there. But that will take a week at least, I am away from home for a while.

Virtex7, 2017-02-20:
ok.
then i will speed it up a bit and start my pull request when the schematic/layout work is done.
i thought about an older kicad version because of the old schematic symbols which are not in the current release... you will see it when it's done...

if you need any help regarding schematic/layouting just tell me, i am a far better electronics engineer than software developer :P