pekkaroi/bldc-drive #10: Missing Binaries

pommen, 2017-12-08:
Hello Pekka!

Your project looks really  interesting to me, though I am struggling to get everything sorted with eclipse. Any chance you could add the binaries to the repository. Just flashing a .hex to a breakout board would do wonders to start prototyping.

Cheers
Peter

pekkaroi, 2017-12-08:
That's a good idea. I added them to the firmware directory. There are two, one compiled with option SINUSOID_DRIVE=1 and one with =0. There isn't too much documentation about what are the differences and unfortunately right now I need to go, so I cannot write too much, but SINUSOID_DRIVE=1 means that the motor is driven by three phase sinusoid voltages and the other uses six-step block commutation.

If you are successful in flashing and get a serial connection to it with the GUI or even get some motor spinning, I am very interested in hearing. I help as much as I can, but please don't expect too much, I've been busy with other stuff and I am starting to forget how this piece of code even worked. :)

pommen, 2017-12-08:
That sounds great, Thanks!
First step is to make some crude proto and try to get a motor spinning, I will keep you updated on my progress.

Best regards
Peter

lonelymyp, 2017-12-08:
I used this manual to set up eclipse https://www.carminenoviello.com/2014/12/28/setting-gcceclipse-toolchain-stm32nucleo-part-1/
I was able to compile, upload and get communication with the controller.