pekkaroi/bldc-drive #12: NOOB guide /Help

Wintertod, 2018-05-08:
can you tell me and maybe others how to use this? how does configuring the servo work? I cant start the GUI when opening with python.

tecfacet, 2018-06-19:
With field oriented control, don't you have to change the commutation tables to sine?  I wish I knew what it would take to change to sine.  I would like to compare sine to trapezoid.

pekkaroi, 2018-06-19:
I have implemented sinusoid drive in this firmware even though not very carefully tested it. Selection between sinusoid or trapezoid is made in file configuration.h. Sine commutation doesn't use hall sensors at all, only encoder to generate the commutation voltages. pwm.c contains the actual implementation of the commutation, that could be a starting point for studying the differences between the methods.

tecfacet, 2018-06-19:
I looked at the code and saw the sine calls.  I will study the code.  On another note, you have mentioned that you recognize the shortfalls of your design..  Can you share some things with me that you feel are not working well or sub standard compared to a commercial servo amp?  I might be able to contribute some code to you.

tecfacet, 2018-06-19:
I also was thinking about using a different method to tune the drive.  Where in the code is the message payload for the tuning software.  I should find this myself but I am taking advantage of you  lol

tecfacet, 2018-06-19:
I have one last question. ... What is the development environment?  Will the software compile under gcc and eclipse?    Thank you.

Wintertod, 2018-06-19:
I had real problems with that too. IIrc he mentioned somewhere that he used eclipse, but a older version. Its a real pain to install the toolchains and stuff. Now I program stm32f103 with the arduino ide. Do you know a ide which is good with stm chips?

tecfacet, 2018-06-19:
I am going to install a newer version of eclipse and gcc.  I used both at Honeywell for Arm cores.  I will let you know how I make out.

Wintertod, 2018-06-19:
So sou know how Good oh performs with dc Motors? I've seen the position accuracy is around 10 encoder steps. Do you have any jmc servos to compare to? And do you know a better alternative for dc servos, like elm Chan?

tecfacet, 2018-09-24:
Hello Again.
Do you have any spare boards for your servo amp?
I am working on some code and would like to try it on your design.
Thank you,
Roger

pekkaroi, 2018-09-27:
Hello. I don't have any finished spare boards. I probably have some bare PCBs somewhere, but quite frankly, shipping the boards from here is probably more expensive than ordering new from China.. And also, if you are willing to try any sort of current control (and I think you should), my board design is not very good. I realized that as the big filtering capacitors are after the current sense resistor, the current sensor is low pass filtered very heavily and thus not suitable for using in the control loop. New board design should be done, but I havent found time to complete it..

tecfacet, 2018-09-27:
I was thinking about doing a FOC design with dual current sense resistors.
What kind of performance are you seeing with your router?
Can you maintain a decent accuracy?  What is lacking, if anything?
Thank you,
Roger

pekkaroi, 2018-09-27:
Very good - dual sensing is definitely the way to go. And FOC. I'd like to implement that some day as well.

In my router, the performance is rather good. I have it in closed loop control with LinuxCNC and I'd say that the accuracy of the motor control stays within 0.01mm during all movements. That is way more than the accuracy of my flexy DIY router. The motors are some Technic motors with 4000cnt/rev encoders.

I then tried to use the drive with my 3d printer and brought some cheap BLDC motors in Nema17 size. I simply was not able to tune the drive to work pretty much at all.. That's pretty much where I'm left at, but haven't got time to investigate further.

Wintertod, 2018-09-27:
For 3d printing I would go with DC motors, especially the 775 60W and 895
360W motors work really good. They go from 3500 to 12000rpm at 36v. I
dropped the whole all in one servo drive concept, there are ready made
bts7960 h bridges of wich 2 get you 4 phase (3 for bldc). Form factor of
the 775 3500 rpm motor is size wise compatible with nema17, i testet 2kg on
a 20t gt2 pulley vertical load. My encoder is a as5040 with 512/1024
interpolated steps. I could tune it to +-1 step accuracy which is
sufficient for 3d printing ffm. Bts79xx are really neat they aee complete
half bridges with current sens and protection at 43A up to 90A at 36v, 2,6€
a piece when you want to build your own 3 phase drive. But really the h
bridge in 43a is on eBay for 5€ which is cheaper than the bts ics. BTW
everything under a certain power I wouldn't recommend bldc,they are too
expensive or not good as a servo.

Another question, have you guys ever thought about implementing sin/cos
encoders? They are insane, I mean a 100s/r encoder interpolated at 10bit
already has 100000ppr, I have a project at work where I have a 1024 sine
/revolution interpolated at 16bit with a native resolution of 67108864ppr.
I thought that we could build them in (low quality) relatively easy, I mean
at 50 or 100 lines with led photodiode and interpolated with 10bit that's
50 to 100k pulses/revolution.

Right now my cost for a 20Ncm servo system is 21€, 7€ the 775 motor 5€ the
h bridge 2€ the stm32f103 and 5€ the as5040 2€ a pcb with headers and LEDs

A rather long reply... On the pictures you see the motor, the 4 holes are
the nema17 holes.

tecfacet, 2018-09-27:
Can you send or give me a picture of how you are using Linuxcnc for tuning?  

I will make PCBs if I can get them to work with LinuxCNC and PMSM.

Can someone post a flowchart or picture?

Thank you,

tecfacet, 2018-09-27:
Wintertod ...
Can you give me more information on your servo system?
What firmware?  How are you tuning?
Thank you,
Roger

Wintertod, 2018-09-27:
Sadly I can't upload pictures or files. Will Try to upload on drive and post link. I'm not tuning with Linux cnc. The servo is controlled via step/dir. You tune the system and then just run it. A problem is the adaptive load. My email is my username  1991@gmail.com, I won't post email for bots reason. Contact me and I send you all my documentation. For discussion I would like to continue on girhub so people can learn from it.

 I will try to upload on zippy share first.

pekkaroi, 2018-09-27:
> Can you send or give me a picture of how you are using Linuxcnc for tuning?
> 
> I will make PCBs if I can get them to work with LinuxCNC and PMSM.
> 
> Can someone post a flowchart or picture?
> 
> Thank you,

I connect my servo drive to my custom Ethernet interface (a very old and outdated post is here: http://pekka.eu/cnc/). So this way I connect the encoder to LinuxCNC and a PWM output from LinuxCNC to the servo AMP. Tuning is done by PID HAL component in LinuxCNC. The same thing is doable with Mesa hardware in LinuxCNC.

Wintertod, 2018-09-27:
So, took a while. Hope you can use some of this. I've come to the conclusion for servos in nema17 format to use DC motors, they go up to 360w with reasonable price 12€/motor and electronics for 14€ wich is a lot cheaper than commercial servos in this range. For nema23 format take jmc bldc servos 100-180w for ~80€ and upward of 300w build stmbl drive, cost about 100€ a piece for a 400v 3kw drive.
Elm chan DC servo has very good code but old hardware and is in assembler . 
I'm looking to build a cheap, modular servo drive. I'm choosing parts on breakout boards you can buy from China, they are smd, cheap and everyone can get them. There's for example a 30a current sensor I'm using instead 9f the building opamp and stuff. 
Would love for you guys to consider this approach of getting complete components and 'plug n play' them on a board with tht. This will make the board a bit bigger but you can change parts easily and the price is mostly cheaper except you plan ob building more than 100 boards. Most cheap cnc or 3d printer with 15€ dremel are capable of milling through hole pcb, so many people can build the 'mother board' or just use protoboards. 
So long rant... Here's the link 
https://www8.zippyshare.com/v/PlzVrBrB/file.html

tecfacet, 2018-09-28:
The link you provided is full of malware.  Can you send the file to my email?  It is username at yahoo.com.

I am interested in your setup.  Please send me links to the components you use to build a servo especially the bridge board and also the file you are using.  Is it the standard hex file from the repository?  I am interested in your implementation.  I will share my new code.

Wintertod, 2018-09-28:
Well the problem with this software is, it's just not mainstream enough. When you are a standard windows user u don't nessesary have python. It took me 6h to sort every problem out till I had a working ui. 2 I needed a usb to serial interface even though stm has usb. I didn't even get to use pekkarois software in the end because i don't have the time to setup a different work environment from scratch which will take days, at least on the first time.
I know it's a bit harder to search for readily available parts online. But not everyone wants to get a reflow oven an in the end I can get working servos for 80€, no need to invest countless hours and more money to get an experimental product.
Like I said, the bridge is just called bts7960. 
I ordered a 895 dc motor today to test it. 5cm diameter is still in nema 17 format, the 5mm shaft is ideal for pulleys and it got an 5mm 2nd shaft for encoder. Seller says peak torque at 24v 16a is 9,8kg/cm but I calculated more like 5kg/cm torque. The speed of 6000rpm should be good too, since it's important. With my system right now I'm able to archive up to 8m/s. With the upgraded motor I'm planning to archive 10m/s with accuracy of 0,04mm and the capability to move 3-4kg loads. 
Sorry for the blocked file access, first time sharing with gdrive.

tecfacet, 2018-09-28:
There are two parts of the equation.
1.  The UI for tuning -  Is this what you rewrote and sent to me?
2.  The servo code.  I believe this is an Eclipse project.  Thoughts?
Sorry for all of the questions .. I am just trying to get up to speed as quick as possible.
Than you.
By the way, where are you located?  I live in New Zealand but currently I am in New Mexico, USA.

Wintertod, 2018-09-28:
The project is the predecessor of pekkarois. It is as far as I can tell not
good written. Information for it were basically non existent and project is
abbadonend. The author is a guy called Mirai iirc.
I'm located in Germany and am a ee for automation, I usually programm in c#
on visual studio, c in arduino because it's an easy environment to use.
For work I mainly use Siemens plc.
The files I send you are a coocox project, I at least got it to compile...
Sadly Coocox is dead
I never tried bldc controll with this code, I have some bldc motors but
none are suitable for precision control. And as I said for low power
applications I would go with DC motors since they are cheaper, controll is
simpler, there are many available. I don't know I you know jmc servos, you
can Google them, they are good mid range bldc servos with 0,55nm and 300%
5s overload capability. In the high power range, buy a asynchronous motor,
put a encoder on it and get an stmbl drive the code for that one is pretty
good they controll a Mitsubishi robot with it.
In the end what I did on this project is mainly sorting of information and
hardware research and design, of wich I can't share all right now and what
I shared is all that's needed anyway.
What is your goal and your budget? My goal was a modular servo that
everyone can build when they have 30€ and access to ebay/aliexpress, I
wanted a controller where you can connect any incremental encoder and motor
driver to get a working system. The code right now works OK, I can controll
the servo with a step/dir impulse from grbl on a arduino.
Don't know if you can use much of it, but I could assist with hardware
design in solid works, pcb design with eagle (even though I don't want to
design a complex pcb) I probably can design a gui in visual studio.