pekkaroi/bldc-drive #13: DC brushed motor version

kamilkamil1, 2020-10-26:
Hi, I would ask about possibility of implementation code for DC brushed motors. I tried change pwm.c file with pwm_Comutate to simple fixed BH1, BL1 parameters but its not work. Pekka please indicate me direction which I should focuse. 

Thanks a lot 

Kamil.

pekkaroi, 2020-10-28:
Hi,

My first recommendation would be to look for projects that already implement this. First one based on a quick search:
- https://github.com/ottoragam/Tarocco

I haven't used brushed DC motors in servo applications ever, but I think it needs two half bridges, not only one. This is because you want to drive current to both directions, so you need to connect the motor between two of the three outputs and then modify the control so that the PWM generation scheme is changed only when changing the direction of the current. There are probably many ways to drive it, I'd most probably try forcing phase1 high and generating PWM on low side of phase2 when driving forward and forcing phase2 high and generating PWM on phase1 low side when reversing.

kamilkamil1, 2020-11-05:
Thank you for your response Pekka, 
I will change the BH1 BH2 BL1 BL2, but every thing should stay as is ? what about Hall timer and initiation of halls? can I disable this ? 

I have another question regarding gui to set parameters, there are some exe file as aplication for windows ?  

Thank you one more time.