pekkaroi/bldc-drive #14: please tell me what does encoder_poles mean

y5664579, 2021-05-23:
Hi, 
Excellent open source project.I try to run it everything is fine but  I don't know the specific meaning of encoder_poles.
It refers to the motor coil pair？It refers to the encoder subdivision？Or something else？
I read the code, but only in one calculation.Please tell me what it means.
Thanks a lot
y5664579.

pekkaroi, 2021-05-24:
Hi. You are correct, it refers to the number of pole pairs in the motors. The name seems to be stupid.. :)

y5664579, 2021-05-24:
> Hi. You are correct, it refers to the number of pole pairs in the motors. The name seems to be stupid.. :)

------
So encoder_poles = all coils / 3
Is it right?
BTW what does commutation_offset mean?
thx for your reply.

artayasa, 2024-01-11:
> Hi. You are correct, it refers to the number of pole pairs in the motors. The name seems to be stupid.. :)

Hi sir, why the GUI i cannot open sir? i already instal newer phyton sir.